brektrou/rtl8821CU #173: Issue when

saeedtabrizi, 2022-05-19:
Hi 
I have an issue when I try to make and  install the driver for rtl8821CU on Linux kernel `5.17.9` via the following steps:

first of all i run the following command :

`uname -r && lsusb`

 and I see the following output :

```
5.17.9-051709-generic
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
then I try to make driver module via following commands : 

`sudo make`

successfully compiled and then

`sudo make install`

the output is :
```
mkdir -p /lib/modules/5.17.9-051709-generic/kernel/drivers/net/wireless/realtek/rtl8821cu/
install -p -m 644 8821cu.ko  /lib/modules/5.17.9-051709-generic/kernel/drivers/net/wireless/realtek/rtl8821cu/
/sbin/depmod -a 5.17.9-051709-generic
```
and everything is normally .
but when i run the following command :
`sudo modprobe 8821cu`

**I got an error** :

`modprobe: ERROR: could not insert '8821cu': Invalid argument`

Then I  see in the `dmesg` command following output :
```
[  275.104092] 8821cu: disagrees about version of symbol __cfg80211_alloc_event_skb
[  275.104095] 8821cu: Unknown symbol __cfg80211_alloc_event_skb (err -22)
[  275.104103] 8821cu: disagrees about version of symbol wiphy_register
[  275.104104] 8821cu: Unknown symbol wiphy_register (err -22)
[  275.104113] 8821cu: disagrees about version of symbol cfg80211_remain_on_channel_expired
[  275.104113] 8821cu: Unknown symbol cfg80211_remain_on_channel_expired (err -22)
[  275.104130] 8821cu: disagrees about version of symbol cfg80211_vendor_cmd_reply
[  275.104131] 8821cu: Unknown symbol cfg80211_vendor_cmd_reply (err -22)
[  275.104138] 8821cu: disagrees about version of symbol cfg80211_del_sta_sinfo
[  275.104139] 8821cu: Unknown symbol cfg80211_del_sta_sinfo (err -22)
[  275.104146] 8821cu: disagrees about version of symbol wiphy_unregister
[  275.104147] 8821cu: Unknown symbol wiphy_unregister (err -22)
[  275.104171] 8821cu: disagrees about version of symbol cfg80211_ch_switch_notify
[  275.104172] 8821cu: Unknown symbol cfg80211_ch_switch_notify (err -22)
[  275.104181] 8821cu: disagrees about version of symbol ieee80211_get_channel_khz
[  275.104182] 8821cu: Unknown symbol ieee80211_get_channel_khz (err -22)
[  275.104190] 8821cu: disagrees about version of symbol cfg80211_ready_on_channel
[  275.104191] 8821cu: Unknown symbol cfg80211_ready_on_channel (err -22)
[  275.104213] 8821cu: disagrees about version of symbol wiphy_free
[  275.104213] 8821cu: Unknown symbol wiphy_free (err -22)
[  275.104218] 8821cu: disagrees about version of symbol wiphy_new_nm
[  275.104219] 8821cu: Unknown symbol wiphy_new_nm (err -22)
[  275.104223] 8821cu: disagrees about version of symbol cfg80211_get_bss
[  275.104224] 8821cu: Unknown symbol cfg80211_get_bss (err -22)
[  275.104240] 8821cu: disagrees about version of symbol __cfg80211_send_event_skb
[  275.104240] 8821cu: Unknown symbol __cfg80211_send_event_skb (err -22)
[  275.104251] 8821cu: disagrees about version of symbol cfg80211_disconnected
[  275.104251] 8821cu: Unknown symbol cfg80211_disconnected (err -22)
[  275.104256] 8821cu: disagrees about version of symbol cfg80211_michael_mic_failure
[  275.104257] 8821cu: Unknown symbol cfg80211_michael_mic_failure (err -22)
[  275.104265] 8821cu: disagrees about version of symbol cfg80211_ibss_joined
[  275.104265] 8821cu: Unknown symbol cfg80211_ibss_joined (err -22)
[  275.104287] 8821cu: disagrees about version of symbol cfg80211_scan_done
[  275.104288] 8821cu: Unknown symbol cfg80211_scan_done (err -22)
[  275.104295] 8821cu: disagrees about version of symbol cfg80211_roamed
[  275.104296] 8821cu: Unknown symbol cfg80211_roamed (err -22)
[  275.104300] 8821cu: disagrees about version of symbol cfg80211_put_bss
[  275.104301] 8821cu: Unknown symbol cfg80211_put_bss (err -22)
[  275.104311] 8821cu: disagrees about version of symbol cfg80211_connect_done
[  275.104311] 8821cu: Unknown symbol cfg80211_connect_done (err -22)
[  275.104323] 8821cu: disagrees about version of symbol cfg80211_unlink_bss
[  275.104324] 8821cu: Unknown symbol cfg80211_unlink_bss (err -22)
[  275.104334] 8821cu: disagrees about version of symbol cfg80211_inform_bss_frame_data
[  275.104335] 8821cu: Unknown symbol cfg80211_inform_bss_frame_data (err -22)
[  275.104343] 8821cu: disagrees about version of symbol cfg80211_new_sta
[  275.104343] 8821cu: Unknown symbol cfg80211_new_sta (err -22)
[  275.104355] 8821cu: disagrees about version of symbol __cfg80211_alloc_reply_skb
[  275.104356] 8821cu: Unknown symbol __cfg80211_alloc_reply_skb (err -22)
[  275.104362] 8821cu: disagrees about version of symbol wiphy_apply_custom_regulatory
[  275.104363] 8821cu: Unknown symbol wiphy_apply_custom_regulatory (err -22)
[  275.104369] 8821cu: disagrees about version of symbol cfg80211_mgmt_tx_status
[  275.104369] 8821cu: Unknown symbol cfg80211_mgmt_tx_status (err -22)
[  275.104374] 8821cu: disagrees about version of symbol cfg80211_rx_mgmt_khz
[  275.104375] 8821cu: Unknown symbol cfg80211_rx_mgmt_khz (err -22)

``` 
and when i try to use dkms :
```
dkms status
nvidia/470.129.06, 5.15.0-30-generic, x86_64: installed
nvidia/470.129.06, 5.17.9-051709-generic, x86_64: installed
rtl8812au/4.3.14: added

```

Then I use the following command to install via dkms :

```
ls
8821cu.ko     8821cu.mod.o   clean   dkms.conf        hal          Kconfig   modules.order   platform     runwpa
8821cu.mod    8821cu.o       core    dkms-install.sh  ifcfg-wlan0  LICENSE   Module.symvers  README.md    wlan0dhcp
8821cu.mod.c  changelog.pdf  debian  dkms-remove.sh   include      Makefile  os_dep          rtl8821c.mk
saeed@pc-912:~/Downloads/build/rtl8821CU$ sudo ./dkms-install.sh 
About to run dkms install steps...
Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source -> /usr/src/rtl8821CU-5.4.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' KVER=5.17.9-051709-generic...............................
Signing module:
 - /var/lib/dkms/rtl8821CU/5.4.1/5.17.9-051709-generic/x86_64/module/8821cu.ko
Secure Boot not enabled on this system.
cleaning build area...

8821cu.ko:
Running module version sanity check.

Good news! Module version v5.4.1_28754.20180921_COEX20180712-3232 for 8821cu.ko
exactly matches what is already found in kernel 5.17.9-051709-generic.
DKMS will not replace this module.
You may override by specifying --force.

depmod...
Finished running dkms install steps.

```
but i got the following output and my **USB wifi dongle not working after reboot** .

```
dkms status
nvidia/470.129.06, 5.15.0-30-generic, x86_64: installed
nvidia/470.129.06, 5.17.9-051709-generic, x86_64: installed
rtl8812au/4.3.14: added
rtl8821CU/5.4.1, 5.17.9-051709-generic, x86_64: installed (WARNING! Diff between built and installed module!)

```

Let me know is there any workaround  or fix for this issue  ?
Thanks

multikirin, 2022-05-21:
hello,

what is your linux distribution? can you send the output of 'cat /etc/os-release'?

Also, try installing cfg80211.

saeedtabrizi, 2022-05-22:
Hi @multikirin 
```
cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
```
Let me know how to  install cfg80211 ?

Thanks .

saeedtabrizi, 2022-07-04:
Hi everyone ,
After I upgrade my Linux kernel to `5.18.3` and using https://github.com/brektrou/rtl8821CU/pull/168 this merge changes to compile and use new module , the problem has been resolved and everything works **FINE**.
I close this issue and thanks to @multikirin contribution.

closed 2022-07-04 by saeedtabrizi