brektrou/rtl8821CU #174: build failed on kernel 5.18

amazingfate, 2022-05-26:
armbian/build#3814

prabhjotsinghchadha, 2022-06-14:
Can someone please fix this issue

mikevmk, 2022-06-20:
$ uname -a
Linux delorean 5.18.0-1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.2-1 (2022-06-06) x86_64 GNU/Linux

a lot of messages about "multiple definition" is_multicast_mac_addr/is_broadcast_mac_addr/is_zero_mac_addr

ld: /home/mike/tmp/rtl8821CU/core/rtw_mp.o: in function `is_multicast_mac_addr':
/home/mike/tmp/rtl8821CU/include/ieee80211.h:1533: multiple definition of `is_multicast_mac_addr'; /home/mike/tmp/rtl8821CU/core/rtw_cmd.o:/home/mike/tmp/rtl8821CU/include/ieee80211.h:1533: first defined here
ld: /home/mike/tmp/rtl8821CU/core/rtw_mp.o: in function `is_broadcast_mac_addr':
/home/mike/tmp/rtl8821CU/include/ieee80211.h:1538: multiple definition of `is_broadcast_mac_addr'; /home/mike/tmp/rtl8821CU/core/rtw_cmd.o:/home/mike/tmp/rtl8821CU/include/ieee80211.h:1538: first defined here
ld: /home/mike/tmp/rtl8821CU/core/rtw_mp.o: in function `is_zero_mac_addr':
/home/mike/tmp/rtl8821CU/include/ieee80211.h:1544: multiple definition of `is_zero_mac_addr'; /home/mike/tmp/rtl8821CU/core/rtw_cmd.o:/home/mike/tmp/rtl8821CU/include/ieee80211.h:1544: first defined here
make[2]: *** [/usr/src/linux-headers-5.18.0-1-common/scripts/Makefile.build:489: /home/mike/tmp/rtl8821CU/8821cu.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-1-common/Makefile:1858: /home/mike/tmp/rtl8821CU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-1-amd64'
make: *** [Makefile:2217: modules] Error 2

mikevmk, 2022-06-20:
This fixed issue for me (took from 41532e3b16dcf27f06e6fe5a26314f3aa24d4f87  https://github.com/aircrack-ng/rtl8812au)

```
--- include/ieee80211.h.orig	2022-06-20 09:52:26.920466272 +0400
+++ include/ieee80211.h	2022-06-20 09:53:01.476963306 +0400
@@ -1529,18 +1529,18 @@
 	(((Addr[2]) & 0xff) == 0xff) && (((Addr[3]) & 0xff) == 0xff) && (((Addr[4]) & 0xff) == 0xff) && \
 				     (((Addr[5]) & 0xff) == 0xff))
 #else
-extern __inline int is_multicast_mac_addr(const u8 *addr)
+static inline int is_multicast_mac_addr(const u8 *addr)
 {
 	return (addr[0] != 0xff) && (0x01 & addr[0]);
 }
 
-extern __inline int is_broadcast_mac_addr(const u8 *addr)
+static inline int is_broadcast_mac_addr(const u8 *addr)
 {
 	return ((addr[0] == 0xff) && (addr[1] == 0xff) && (addr[2] == 0xff) &&   \
 		(addr[3] == 0xff) && (addr[4] == 0xff) && (addr[5] == 0xff));
 }
 
-extern __inline int is_zero_mac_addr(const u8 *addr)
+static inline int is_zero_mac_addr(const u8 *addr)
 {
 	return ((addr[0] == 0x00) && (addr[1] == 0x00) && (addr[2] == 0x00) &&   \
 		(addr[3] == 0x00) && (addr[4] == 0x00) && (addr[5] == 0x00));

```

guillaje, 2022-06-24:
The patch found by [mikevmk](https://github.com/mikevmk) worked for me too.

prabhjotsinghchadha, 2022-06-24:
I don't have much idea, where to make these code changes? for me, it's still not working.. @guillaje @mikevmk

birdofprey, 2022-06-28:
> I don't have much idea, where to make these code changes? for me, it's still not working.. @guillaje @mikevmk

os:Debian Gnu/Linux
kernel:5.18.0-0.bpo.1-amd64 (x86_64)
error:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1533: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl882
1CU/5.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1533: first defined here
ld: /var/lib/dkms/rtl8821CU/5.4.1/build/hal/btc/halbtc8821c2ant.o: in function `is_broadcast_mac_addr':
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1538: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl882
1CU/5.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1538: first defined here
ld: /var/lib/dkms/rtl8821CU/5.4.1/build/hal/btc/halbtc8821c2ant.o: in function `is_zero_mac_addr':
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1544: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl8821CU/5
.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1544: first defined here
ld: /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mp.o: in function `is_multicast_mac_addr':
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1533: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl882
1CU/5.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1533: first defined here
ld: /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mp.o: in function `is_broadcast_mac_addr':
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1538: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl882
1CU/5.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1538: first defined here
ld: /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mp.o: in function `is_zero_mac_addr':
/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1544: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl8821CU/5
.4.1/build/core/rtw_cmd.o:/var/lib/dkms/rtl8821CU/5.4.1/build/include/ieee80211.h:1544: first defined here
make[2]: *** [/usr/src/linux-headers-5.18.0-0.bpo.1-common/scripts/Makefile.build:489：/var/lib/dkms/rtl8821CU/5.4.1/build/8821cu.
o] 错误 1
make[1]: *** [/usr/src/linux-headers-5.18.0-0.bpo.1-common/Makefile:1858：/var/lib/dkms/rtl8821CU/5.4.1/build] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.18.0-0.bpo.1-amd64”
make: *** [Makefile:2217：modules] 错误 2

mkaskov, 2022-06-29:
does not work on Manjaro with kernel 5.18.x. with kernel 5.17.x works fine.
if add changes from https://github.com/brektrou/rtl8821CU/pull/168 it start work with kernel 5.18.x fine.

mkaskov, 2022-06-30:
why did you close this ticket?
this repo does not work with kernel 5.18.x now.
please approve this pull request https://github.com/brektrou/rtl8821CU/pull/168 first

amazingfate, 2022-06-30:
@mkaskov sorry about that, I created this issue because armbian build failed. I found the build is successful yesterday then I think this issue should get solved by upstream. But I recheck the code today and find that armbian is using a patch to fix this issue, so I reopen it.