brektrou/rtl8821CU #176: not work on ubuntu18.04

koalabearguo, 2022-06-13:
```
dmesfg
[    7.767767] RTW: module init start
[    7.767769] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[    7.767769] RTW: build time: Jun 13 2022 21:08:51
[    7.767770] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[    7.770126] mc: Linux media interface: v0.10
[    7.777533] videodev: Linux video capture interface: v2.00
[    7.889897] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[    7.890594] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    7.898683] usbcore: registered new interface driver btusb
[    7.901615] Bluetooth: hci0: Firmware revision 0.3 build 250 week 9 2022
[    7.916944] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (1bcf:28b0)
[    7.918242] uvcvideo: Failed to query (GET_INFO) UVC control 3 on unit 1: -32 (exp. 1).
[    7.939474] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input25
[    7.940115] usbcore: registered new interface driver uvcvideo
[    7.940116] USB Video Class driver (1.1.1)
[    7.968677] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.968678] Bluetooth: BNEP filters: protocol multicast
[    7.968681] Bluetooth: BNEP socket layer initialized
[    7.988793] EXT4-fs (sda5): re-mounted. Opts: errors=remount-ro,commit=360
[    7.998983] RTW: HW EFUSE
[    7.998985] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[    7.998990] RTW: 0x010: 20 20 20 22  22 22 26 26  26 28 28 04  FF FF FF FF  
[    7.998994] RTW: 0x020: FF FF 26 25  24 23 22 22  23 23 24 24  23 21 21 21  
[    7.998998] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 20 20  20 22 22 22  
[    7.999003] RTW: 0x040: 26 26 26 28  28 04 FF FF  FF FF FF FF  FF FF FF FF  
[    7.999007] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999011] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999015] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999020] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999024] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999028] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999032] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 34 1D 00  FF FF FF FF  
[    7.999037] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[    7.999041] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999045] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999049] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999054] RTW: 0x100: DA 0B 11 C8  E3 46 02 E0  E1 A9 1D A7  E5 09 03 52  
[    7.999058] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[    7.999062] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[    7.999066] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999071] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[    7.999075] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999079] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999083] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999087] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999092] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999096] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999100] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999104] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999109] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999113] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999117] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[    7.999189] RTW: hal_com_config_channel_plan chplan:0x7F
[    8.004670] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[    8.005880] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[    8.052559] EXT4-fs (sda6): re-mounted. Opts: commit=360
[    8.058721] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.058818] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.058875] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.058965] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.059071] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.059135] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.059214] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    8.059501] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[    8.060684] RTW: rtw_ndev_init(wlan1) if1 mac_addr=e0:e1:a9:1d:a7:e5
[    8.061124] usbcore: registered new interface driver rtl8821cu
[    8.061125] RTW: module init ret=0

```
```
uname -a
Linux koala-pc 5.4.0-117-generic #132~18.04.1-Ubuntu SMP Thu Jun 2 23:36:48 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
```
cat /etc/os-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

```

uncleAntik, 2022-06-27:
Same issue on Fedora 36 5.18.x kernel. Need some help over here

troyfolger, 2022-06-29:
The developer of this project is keeping the driver updated for new kernels:
https://github.com/morrownr/8821cu-20210118

yanggis, 2022-07-25:
I'm on the same boat, any suggestion?