brektrou/rtl8821CU #183: "Make" error. Also have errors when using dkms

kappaisagreekletter, 2022-07-25:
Errors using Make:
[image]
[image]
[image]

Errors using dkms:
[image]

Make log:
[make.log](https://github.com/brektrou/rtl8821CU/files/9179737/make.log)

CMD-Josh, 2022-07-26:
Had a similar experience,
Upgraded from kernel 5.15.11-arch2-1 to 5.18.14-arch1-1
Adaptor no longer works. It shows up perfectly fine when I lsusb but it is not recognised as a wireless interface.

Tried installing drivers manually through Make and DKMS but met with the same errors as kappaisagreekletter

Make file output: 
[MakeOutput.txt](https://github.com/brektrou/rtl8821CU/files/9191911/MakeOutput.txt)

DKMS file output:
[DkmsOutput.txt](https://github.com/brektrou/rtl8821CU/files/9191913/DkmsOutput.txt)

kappaisagreekletter, 2022-07-27:
I reverted to my old kernel and the driver for my wifi USB works again.
https://www.reddit.com/r/pop_os/comments/rbu53y/reverting_the_kernel_version/

kul-sudo, 2022-10-18:
Any fix so far?

kul-sudo, 2022-10-18:
> I reverted to my old kernel and the driver for my wifi USB works again. https://www.reddit.com/r/pop_os/comments/rbu53y/reverting_the_kernel_version/
> 
> You can also manually install an old kernel: https://itsfoss.com/upgrade-linux-kernel-ubuntu/#install-manually

But how exactly did you go to an old kernel?

kappaisagreekletter, 2022-10-18:
> But how exactly did you go to an old kernel?

follow this guide:
[itsfoss.com/upgrade-linux-kernel-ubuntu/#install-manually](https://itsfoss.com/upgrade-linux-kernel-ubuntu/#install-manually)