brektrou/rtl8821CU #184: [Solution] How to build on raspberry pi.

zoldaten, 2022-08-16:
as i crashed many times with fail in kernel-headers here is the solution how to fix it:
```
sudo apt-get update
sudo apt-get install dkms
sudo apt-get update && sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
sudo apt-get install raspberrypi-kernel-headers
reboot
git clone https://github.com/brektrou/rtl8821CU.git
cd rtl8821CU
sudo ./dkms-install.sh
```
work on rasberry pi arm7l.

have a nice day!

Sofronio, 2022-08-24:
It works! Thanks!

matuzalemmuller, 2022-09-25:
You, sir, are a gentleman and a scholar. Confirmed working on Rasp 3B+.

bdwilson, 2022-09-29:
don't forget apt-get install bc

actraiser30, 2023-05-17:
Hi its not working in my raspberry pi 2 in latest raspbian bullseye