brektrou/rtl8821CU #193: Incompatable with 6.0.2 Kernel

NanzeRT, 2022-10-18:
New Kernel changed API so driver wouldn't compile on it.

`make` output:
```make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.2-zen1-1-zen/build M=/home/nanzert/Repos/Other/rtl8821CU  modules
make[1]: Entering directory '/usr/lib/modules/6.0.2-zen1-1-zen/build'
  CC [M]  /home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.o
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:420:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  420 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/nanzert/Repos/Other/rtl8821CU/include/osdep_service_linux.h:93,
                 from /home/nanzert/Repos/Other/rtl8821CU/include/osdep_service.h:50,
                 from /home/nanzert/Repos/Other/rtl8821CU/include/drv_types.h:27,
                 from /home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8261:6: note: declared here
 8261 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:1088:26: error: ‘struct cfg80211_roam_info’ has no member named ‘bssid’
 1088 |                 roam_info.bssid = cur_network->network.MacAddress;
      |                          ^
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_scan’:
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:2867:33: warning: the comparison will always evaluate as ‘true’ for the address of ‘ssid’ will never be NULL [-Waddress]
 2867 |                 if (ssids->ssid != NULL
      |                                 ^~
./include/net/cfg80211.h:2357:12: note: ‘ssid’ declared here
 2357 |         u8 ssid[IEEE80211_MAX_SSID_LEN];
      |            ^~~~
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
 9416 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9682:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
 9682 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /home/nanzert/Repos/Other/rtl8821CU/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1849: /home/nanzert/Repos/Other/rtl8821CU] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.2-zen1-1-zen/build'
make: *** [Makefile:2217: modules] Error 2
```

Electrominch, 2022-10-18:
Same problem

kelebek333, 2022-10-20:
You can use that.   https://github.com/morrownr/8821cu-20210118

Electrominch, 2022-10-20:
https://github.com/morrownr/8821cu-20210118/issues/52

Electrominch, 2022-10-22:
It works! Thx

breisig, 2022-11-04:
same problem. Can we get this fixed  in the code?