brektrou/rtl8821CU #194: Issue with kernel 5.19.16 on pop_os

cdhassell, 2022-10-20:
dan@localhost:~/build/rtl8821CU$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.16-76051916-generic/build M=/home/dan/build/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.16-76051916-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_cmd.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_security.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_debug.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_io.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_ioctl_query.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_ioctl_set.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_ieee80211.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_mlme.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_mlme_ext.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_mi.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_wlan_util.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_vht.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_pwrctrl.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_rf.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_recv.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_sta_mgt.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_ap.o
  CC [M]  /home/dan/build/rtl8821CU/core/mesh/rtw_mesh.o
  CC [M]  /home/dan/build/rtl8821CU/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/dan/build/rtl8821CU/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_xmit.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_p2p.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_rson.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_tdls.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_br_ext.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_iol.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_sreset.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_btcoex_wifionly.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_btcoex.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_beamforming.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_odm.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_rm.o
  CC [M]  /home/dan/build/rtl8821CU/core/rtw_rm_fsm.o
  CC [M]  /home/dan/build/rtl8821CU/core/efuse/rtw_efuse.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/osdep_service.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/os_intfs.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/usb_intf.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/xmit_linux.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/mlme_linux.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/recv_linux.o
  CC [M]  /home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.o
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:420:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  420 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/dan/build/rtl8821CU/include/osdep_service_linux.h:93,
                 from /home/dan/build/rtl8821CU/include/osdep_service.h:50,
                 from /home/dan/build/rtl8821CU/include/drv_types.h:27,
                 from /home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7967:6: note: declared here
 7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
 9416 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9682:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
 9682 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /home/dan/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1852: /home/dan/build/rtl8821CU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.16-76051916-generic'
make: *** [Makefile:2217: modules] Error 2

netlore, 2022-10-21:
Same issue here on Ubuntu with 5.19.0-21, it seems there was some refactoring of the networking in general as well as WiFi in 5.19.

intropedro, 2022-11-25:
With the fork https://github.com/ylxdzsw/rtl8821CU it works