brektrou/rtl8821CU #206: 8821cu on Ubuntu 22.04 5.19.0-32 ERROR

leongithub, 2023-02-12:
> sudo ./dkms-install.sh
> 
> About to run dkms install steps...
> Error! DKMS tree already contains: rtl8821CU-5.4.1
> You cannot add the same module/version combo more than once.
> 
> Kernel preparation unnecessary for this kernel. Skipping...
> 
> Building module:
> cleaning build area...
> 'make' KVER=5.19.0-32-generic....................................(bad exit status: 2)
> ERROR (dkms apport): binary package for rtl8821CU: 5.4.1 not found
> Error! Bad return status for module build on kernel: 5.19.0-32-generic (aarch64)
> Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
> 
> Kernel preparation unnecessary for this kernel. Skipping...
> 
> Building module:
> cleaning build area...
> 'make' KVER=5.19.0-32-generic....................................(bad exit status: 2)
> ERROR (dkms apport): binary package for rtl8821CU: 5.4.1 not found
> Error! Bad return status for module build on kernel: 5.19.0-32-generic (aarch64)
> Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
> Finished running dkms install steps.

/var/lib/dkms/rtl8821CU/5.4.1/build/make.log

> DKMS make.log for rtl8821CU-5.4.1 for kernel 5.19.0-32-generic (aarch64)
> 2023年 02月 12日 星期日 09:31:39 CST
> make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.19.0-32-generic/build M=/var/lib/dkms/rtl8821CU/5.4.1/build  modules
> make[1]: 进入目录“/usr/src/linux-headers-5.19.0-32-generic”
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
>   You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_io.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_query.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_set.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ieee80211.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme_ext.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mi.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_wlan_util.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_vht.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_pwrctrl.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rf.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_recv.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_sta_mgt.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ap.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh_pathtbl.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh_hwmp.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_xmit.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_p2p.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rson.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_tdls.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_br_ext.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_iol.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_sreset.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_btcoex_wifionly.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_btcoex.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_beamforming.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_odm.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm_fsm.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2494:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
>  2494 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
>       |                     ~~~~~~~^~~~~~~~~~
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:925:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
>   925 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
>       |                  ~~~~~~^~~
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/os_intfs.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/usb_intf.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/usb_ops_linux.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_linux.o
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_linux.c: In function ‘rtw_mp_efuse_set’:
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_linux.c:9781:46: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
>  9781 |                 _rtw_memcpy(padapter->pnetdev->dev_addr, get_hal_mac_addr(padapter), ETH_ALEN); /* set mac addr to net_device */
>       |                             ~~~~~~~~~~~~~~~~~^~~~~~~~~~
> In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
>                  from /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_linux.c:17:
> /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:296:35: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
>   296 | extern void     _rtw_memcpy(void *dec, const void *sour, u32 sz);
>       |                             ~~~~~~^~~
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/xmit_linux.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/mlme_linux.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/recv_linux.o
>   CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.o
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c:420:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
>   420 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
>       |         ^~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:93,
>                  from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
>                  from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
>                  from /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c:17:
> ./include/net/cfg80211.h:7967:6: note: declared here
>  7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
>       |      ^~~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c:9416:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
>  9416 |         .stop_ap = cfg80211_rtw_stop_ap,
>       |                    ^~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c:9416:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
> /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.c:9682:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
>  9682 |         if (wdev->current_bss) {
>       |                 ^~
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:257：/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/linux/ioctl_cfg80211.o] 错误 1
> make[1]: *** [Makefile:1850：/var/lib/dkms/rtl8821CU/5.4.1/build] 错误 2
> make[1]: 离开目录“/usr/src/linux-headers-5.19.0-32-generic”
> make: *** [Makefile:2217：modules] 错误 2

I recently upgraded the kernel version and reinstalled it and it went wrong. ask for help

Terminator637, 2023-02-16:
Hi! Got the same trouble. Ubuntu 22.04 kernel 5.19.0-32-generic

jfischoff, 2023-02-17:
same problem

bayeswhu, 2023-02-18:
> Hi! Got the same trouble. Ubuntu 22.04 kernel 5.19.0-32-generic

same here

Abathi, 2023-02-18:
Got the same error!

johnnyyan, 2023-02-18:
Is there a workaround?

yujiang0030, 2023-02-18:
Hi, I had got the same error. Then I used this driver https://github.com/morrownr/8821cu-20210916 to replace the old one, and the network recoverd.

Dephrilibrium, 2023-02-18:
> Hi, I had got the same error. Then I used this driver https://github.com/morrownr/8821cu-20210916 to replace the old one, and the network recoverd.

Can confirm, this one compiles without errors and works afterwards.

But I have one additional question, someone may know to fix.

Very often my WiFi won't work anymore, when I reboot after an update/upgrade my Ubuntu 22.04 LTS by:
`sudo apt update & sudo apt upgrade`
Then the WiFi-Adapter won't work anymore until I completely recompile "reinstall" it (only uninstall and install without recompiling won't work).

I wrote some automated script to do that for me, so that I don't have to input the commands all the time by hand:
```
#!/bin/bash
# Change into sourcecode directory
#cd ~/Downloads/rtl8821cu/            # Previous repo (not working currently)
cd ~/Downloads/8821cu-20210916       # Current working repo
# Grab changes if there are some
git pull
sudo make clean
sudo make
sudo make uninstall
sudo make install
```
But that is soooo annoying, because I need to wait that the driver got recompiled and reinstalled -_-.
So, has anyone the same problem and know how to avoid that behaviour?

Kind regards,
Deph

jfischoff, 2023-02-19:
@Dephrilibrium's steps worked for me

nosun, 2023-02-19:
@Dephrilibrium's steps worked for me

And may be add more command
```
sudo modprobe 8821cu
```

Dephrilibrium, 2023-02-19:
> @Dephrilibrium's steps worked for me
> 
> And maybe add another command to make it work
> 
> ```
> sudo modprobe 8821cu
> ```

Thanks for the hint.
Personally I use the auto modeswitch described by the author in the repositories [README.md](https://github.com/brektrou/rtl8821CU#make-it-permanent), but I will add it in a second and also provide my shortcut, so that it can be called from the start-menu.

Kind regards,
Deph

braunerdev, 2023-02-27:
@Dephrilibrium's steps worked for me
Thank you Bro! no more cables to work hehehe

scriptkid2222, 2023-03-11:
> Olá, eu tive o mesmo erro. Então usei este driver https://github.com/morrownr/8821cu-20210916 para substituir o antigo e a rede recuperou.

this one worked

RyanMan56, 2023-03-11:
> Hi, I had got the same error. Then I used this driver https://github.com/morrownr/8821cu-20210916 to replace the old one, and the network recoverd.

Cheers this worked for me :)

bensontan, 2023-05-30:
rtl8821CU/8821cu-20210916/hal/hal_com.c:13781:17: error: too many arguments to function ‘rtw_halmac_bf_cfg_csi_rate’
13781 |                 rtw_halmac_bf_cfg_csi_rate(adapter_to_dvobj(adapter),
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/benson/rtl8821CU/8821cu-20210916/hal/hal_com.c:23:
/home/benson/rtl8821CU/8821cu-20210916/hal/../../hal/hal_halmac.h:229:5: note: declared here
  229 | int rtw_halmac_bf_cfg_csi_rate(struct dvobj_priv *d, u8 rssi, u8 current_rate,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~

comment out as following in hal_com.c: #13781
		rtw_halmac_bf_cfg_csi_rate(adapter_to_dvobj(adapter),
				p_dm_odm->rssi_min,
				bf_info->cur_csi_rpt_rate,
				fix_rate_enable, &new_csi_rate_idx); // , &rrsr_54_en);

		temp_rrsr = rtw_read32(adapter, REG_RRSR);
		// if (rrsr_54_en == 1)
		// 	temp_rrsr |= RRSR_54M;
		// else if (rrsr_54_en == 0)
		// 	temp_rrsr &= ~RRSR_54M;
		rtw_phydm_set_rrsr(adapter, temp_rrsr, FALSE);

Dephrilibrium, 2023-08-05:
Heyho everyone, 

yesterday I've made an update/upgrade of my ubuntu 22.04 LTS. Thereby, the latest kernel (6.2.0-26-generic) was installed, and my dongle showed no WiFi anymore in the taskbar network-window.

After playing around and a lot of restarts and recompile-tries, I came up with the idea, that linux is may confused by the different drivers, still stored within the different kernel-version folders in  `/lib/modules/` (see Makefile: `MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/`).

Therefore I decided to delete all my older kernel version folders (`sudo rm -r 5*`), restarted the PC and recompiled with my script above https://github.com/brektrou/rtl8821CU/issues/206#issuecomment-1435713760. One restart later, everything worked well again.

Hope that helps, if some is struggling with the same problem.

Deph