brektrou/rtl8821CU #208: /usr/sbin/usb_modeswitch -K -v 0bda -p c811 fails

rwb196884, 2023-02-14:
Any ideas?!

```
Feb 14 21:09:29 mini31 kernel: [29109.764866] usb 1-1: USB disconnect, device number 16
Feb 14 21:09:33 mini31 kernel: [29112.995686] usb 1-1: new high-speed USB device number 17 using ehci-pci
Feb 14 21:09:33 mini31 kernel: [29113.153150] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
Feb 14 21:09:33 mini31 kernel: [29113.153154] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 14 21:09:33 mini31 kernel: [29113.153157] usb 1-1: Product: 802.11ac NIC
Feb 14 21:09:33 mini31 kernel: [29113.153159] usb 1-1: Manufacturer: Realtek
Feb 14 21:09:33 mini31 kernel: [29113.153161] usb 1-1: SerialNumber: 123456
Feb 14 21:09:33 mini31 systemd-udevd[1704]: Process '/usr/sbin/usb_modeswitch -K -v 0bda -p c811' failed with exit code 1.
Feb 14 21:09:33 mini31 kernel: [29113.207904] RTW: module init start
Feb 14 21:09:33 mini31 kernel: [29113.207907] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
Feb 14 21:09:33 mini31 kernel: [29113.207907] RTW: build time: Feb 14 2023 21:03:30
Feb 14 21:09:33 mini31 kernel: [29113.207908] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
Feb 14 21:09:33 mini31 kernel: [29113.208006] RTW: [HALMAC]11692M
Feb 14 21:09:33 mini31 kernel: [29113.208006] HALMAC_MAJOR_VER = 1
Feb 14 21:09:33 mini31 kernel: [29113.208006] HALMAC_PROTOTYPE_VER = 4
Feb 14 21:09:33 mini31 kernel: [29113.208006] HALMAC_MINOR_VER = 19
Feb 14 21:09:33 mini31 kernel: [29113.208006] HALMAC_PATCH_VER = 3
Feb 14 21:09:33 mini31 kernel: [29113.209558] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:09:33 mini31 kernel: [29113.485926] RTW: HW EFUSE
Feb 14 21:09:33 mini31 kernel: [29113.485929] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B
Feb 14 21:09:33 mini31 kernel: [29113.485939] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.485947] RTW: 0x020: FF FF 1D 1C  1C 1C 1E 1E  1E 1D 1E 1E  20 20 22 21
Feb 14 21:09:33 mini31 kernel: [29113.485955] RTW: 0x030: F3 FF FF FF  FF FF 00 FF  FF FF 23 22  22 23 23 23
Feb 14 21:09:33 mini31 kernel: [29113.485962] RTW: 0x040: 25 25 25 26  26 10 FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.485970] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.485978] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.485985] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.485993] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486001] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486008] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486016] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 0F 1E 00  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486023] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486031] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486039] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486046] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486054] RTW: 0x100: DA 0B 11 C8  E3 46 02 00  E0 32 80 03  84 09 03 52
Feb 14 21:09:33 mini31 kernel: [29113.486061] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
Feb 14 21:09:33 mini31 kernel: [29113.486069] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486077] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486084] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486092] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486100] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486107] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486115] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486122] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486130] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486138] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486145] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486153] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486160] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486168] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:09:33 mini31 kernel: [29113.486407] RTW: hal_com_config_channel_plan chplan:0x7F
Feb 14 21:09:33 mini31 kernel: [29113.503170] RTW: [HALMAC][ALWAYS]shall R reg twice!!
Feb 14 21:09:33 mini31 kernel: [29113.506913] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
Feb 14 21:09:33 mini31 kernel: [29113.606662] RTW: ERROR [HALMAC][ERR]h2cq compare!!
Feb 14 21:09:33 mini31 kernel: [29113.606665] RTW: ERROR rtl8821c_fw_dl Download Firmware from array failed
Feb 14 21:09:33 mini31 kernel: [29113.751793] usb 1-1: reset high-speed USB device number 17 using ehci-pci
Feb 14 21:09:33 mini31 kernel: [29113.908596] usbcore: registered new interface driver rtl8821cu
Feb 14 21:09:33 mini31 kernel: [29113.908600] RTW: module init ret=0
```

```
# lsusb
Bus 002 Device 005: ID 22b8:2e24 Motorola PCS
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 004 Device 003: ID 05ac:8216 Apple, Inc. Bluetooth USB Host Controller
Bus 004 Device 002: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 017: ID 0bda:c811 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 05ac:8242 Apple, Inc. Built-in IR Receiver
Bus 003 Device 002: ID 0451:16a8 Texas Instruments, Inc.
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

```
 # uname -a
Linux mini31 4.19.0-23-amd64 #1 SMP Debian 4.19.269-1 (2022-12-20) x86_64 GNU/Linux
```
(Debian 10)

rwb196884, 2023-02-14:
Removing and re-inserting the adapter a dozen or so times and got lucky!

```
Feb 14 21:13:13 mini31 kernel: [29333.765494] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 14 21:13:13 mini31 kernel: [29333.765497] usb 1-1: Product: 802.11ac NIC
Feb 14 21:13:13 mini31 kernel: [29333.765499] usb 1-1: Manufacturer: Realtek
Feb 14 21:13:13 mini31 kernel: [29333.765501] usb 1-1: SerialNumber: 123456
Feb 14 21:13:13 mini31 kernel: [29333.766170] RTW: [HALMAC]11692M
Feb 14 21:13:13 mini31 kernel: [29333.766170] HALMAC_MAJOR_VER = 1
Feb 14 21:13:13 mini31 kernel: [29333.766170] HALMAC_PROTOTYPE_VER = 4
Feb 14 21:13:13 mini31 kernel: [29333.766170] HALMAC_MINOR_VER = 19
Feb 14 21:13:13 mini31 kernel: [29333.766170] HALMAC_PATCH_VER = 3
Feb 14 21:13:13 mini31 kernel: [29333.767633] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.049519] RTW: HW EFUSE
Feb 14 21:13:14 mini31 kernel: [29334.049523] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B
Feb 14 21:13:14 mini31 kernel: [29334.049532] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049539] RTW: 0x020: FF FF 1D 1C  1C 1C 1E 1E  1E 1D 1E 1E  20 20 22 21
Feb 14 21:13:14 mini31 kernel: [29334.049546] RTW: 0x030: F3 FF FF FF  FF FF 00 FF  FF FF 23 22  22 23 23 23
Feb 14 21:13:14 mini31 kernel: [29334.049553] RTW: 0x040: 25 25 25 26  26 10 FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049560] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049568] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049575] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049582] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049589] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049596] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049603] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 0F 1E 00  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049610] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049617] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049624] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049631] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049638] RTW: 0x100: DA 0B 11 C8  E3 46 02 00  E0 32 80 03  84 09 03 52
Feb 14 21:13:14 mini31 kernel: [29334.049645] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
Feb 14 21:13:14 mini31 kernel: [29334.049652] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049660] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049667] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049674] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049681] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049688] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049695] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049702] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049709] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049716] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049723] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049730] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049737] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.049745] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Feb 14 21:13:14 mini31 kernel: [29334.050016] RTW: hal_com_config_channel_plan chplan:0x7F
Feb 14 21:13:14 mini31 kernel: [29334.069875] RTW: [HALMAC][ALWAYS]shall R reg twice!!
Feb 14 21:13:14 mini31 kernel: [29334.073750] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
Feb 14 21:13:14 mini31 kernel: [29334.189258] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.189500] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.189757] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.189997] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.190249] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.190497] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.190747] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Feb 14 21:13:14 mini31 kernel: [29334.191444] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[... cut ...]
Feb 14 21:13:14 mini31 kernel: [29334.521411] R10: 00007ffef3e77844 R11: 0000000000000246 R12: 0000558548019870
Feb 14 21:13:14 mini31 kernel: [29334.521413] R13: 00007ffef3e77770 R14: 0000000000000000 R15: 0000558548019df0
Feb 14 21:13:14 mini31 kernel: [29334.521416] ---[ end trace 5c7caa1cd8e252e6 ]---
Feb 14 21:13:14 mini31 vnstatd[782]: Info: Interface "wlx00e032800384" enabled.
Feb 14 21:13:15 mini31 kernel: [29335.421239] RTW: wlx00e032800384- hw port(0) mac_addr =00:e0:32:80:03:84
...
```

x931890193, 2023-07-12:
/

> 

### Same Situation， can not solve！！！

ChienKaiMa, 2023-11-09:
Copy from my comment in [issue #221](https://github.com/brektrou/rtl8821CU/issues/221#issuecomment-1804609648) :
I had the same problem. I saw some comments mentioning the following repo and it worked for me.
[8821cu-20210916](https://github.com/morrownr/8821cu-20210916)