brektrou/rtl8821CU #215: install via dkms error

Robson96, 2023-02-26:
r0b1nh00d@security:~/rtl8821CU$ sudo ./dkms-install.sh
About to run dkms install steps...
Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source -> /usr/src/rtl8821CU-5.4.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' KVER=5.19.0-32-generic..................................................................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8821CU: 5.4.1 not found
Error! Bad return status for module build on kernel: 5.19.0-32-generic (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' KVER=5.19.0-32-generic.......................................................^C

jvvasquesn, 2023-03-02:
Try this
https://github.com/morrownr/8821cu-20210916

Robson96, 2023-03-03:
I installed it, and it worked perfectly.

closed 2023-03-03 by Robson96