brektrou/rtl8821CU #216: Multiple compilation errors on linux 6.1.19

vitek, 2023-03-29:
```
(master) vitja@vitja-laptop:~/work/rtl8821CU$ LC_ALL=C make -k
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.19-gentoo-dist/build M=/home/vitja/work/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-6.1.19-gentoo-dist'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 12.2.1_p20230121-r1 p10) 12.2.1 20230121
  You are using:           gcc (Gentoo 12.2.1_p20230121-r1 p10) 12.2.1 20230121
  CC [M]  /home/vitja/work/rtl8821CU/os_dep/linux/os_intfs.o
/home/vitja/work/rtl8821CU/os_dep/linux/os_intfs.c: В функции «rtw_os_ndev_register»:
/home/vitja/work/rtl8821CU/os_dep/linux/os_intfs.c:1646:9: ошибка: слишком много аргументов в вызове функции «netif_napi_add»
 1646 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /home/vitja/work/rtl8821CU/include/osdep_service_linux.h:30,
                 from /home/vitja/work/rtl8821CU/include/osdep_service.h:50,
                 from /home/vitja/work/rtl8821CU/include/drv_types.h:27,
                 from /home/vitja/work/rtl8821CU/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2569:1: замечание: объявлено здесь
 2569 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /home/vitja/work/rtl8821CU/os_dep/linux/os_intfs.o] Ошибка 1
  CC [M]  /home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.o
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: В функции «rtw_cfg80211_ch_switch_notify»:
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:420:9: ошибка: слишком мало аргументов в вызове функции «cfg80211_ch_switch_notify»
  420 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/vitja/work/rtl8821CU/include/osdep_service_linux.h:93,
                 from /home/vitja/work/rtl8821CU/include/osdep_service.h:50,
                 from /home/vitja/work/rtl8821CU/include/drv_types.h:27,
                 from /home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8276:6: замечание: объявлено здесь
 8276 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: В функции «rtw_cfg80211_indicate_connect»:
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:1088:26: ошибка: «struct cfg80211_roam_info» не содержит элемента с именем «bssid»
 1088 |                 roam_info.bssid = cur_network->network.MacAddress;
      |                          ^
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: В функции «cfg80211_rtw_scan»:
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:2867:33: предупреждение: the comparison will always evaluate as «true» for the address of «ssid» will never be NULL [-Waddress]
 2867 |                 if (ssids->ssid != NULL
      |                                 ^~
./include/net/cfg80211.h:2359:12: замечание: «ssid» объявлено здесь
 2359 |         u8 ssid[IEEE80211_MAX_SSID_LEN];
      |            ^~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: На верхнем уровне:
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9381:20: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, struct key_params *)» {aka «int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, struct key_params *)»} from incompatible pointer type «int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, struct key_params *)» {aka «int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, struct key_params *)»} [-Werror=incompatible-pointer-types]
 9381 |         .add_key = cfg80211_rtw_add_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9381:20: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.add_key»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9382:20: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))» {aka «int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))»} from incompatible pointer type «int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))» {aka «int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))»} [-Werror=incompatible-pointer-types]
 9382 |         .get_key = cfg80211_rtw_get_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9382:20: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.get_key»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9383:20: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *)» {aka «int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *)»} from incompatible pointer type «int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *)» {aka «int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *)»} [-Werror=incompatible-pointer-types]
 9383 |         .del_key = cfg80211_rtw_del_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9383:20: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.del_key»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9384:28: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  bool)» {aka «int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  _Bool)»} from incompatible pointer type «int (*)(struct wiphy *, struct net_device *, u8,  bool,  bool)» {aka «int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  _Bool)»} [-Werror=incompatible-pointer-types]
 9384 |         .set_default_key = cfg80211_rtw_set_default_key,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9384:28: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.set_default_key»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9386:33: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, int,  u8)» {aka «int (*)(struct wiphy *, struct net_device *, int,  unsigned char)»} from incompatible pointer type «int (*)(struct wiphy *, struct net_device *, u8)» {aka «int (*)(struct wiphy *, struct net_device *, unsigned char)»} [-Werror=incompatible-pointer-types]
 9386 |         .set_default_mgmt_key = cfg80211_rtw_set_default_mgmt_key,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9386:33: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.set_default_mgmt_key»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: ошибка: initialization of «int (*)(struct wiphy *, struct net_device *, unsigned int)» from incompatible pointer type «int (*)(struct wiphy *, struct net_device *)» [-Werror=incompatible-pointer-types]
 9416 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9416:20: замечание: (где-то рядом с инициализацией для «rtw_cfg80211_ops.stop_ap»)
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c: В функции «rtw_wdev_unregister»:
/home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.c:9682:17: ошибка: «struct wireless_dev» не содержит элемента с именем «current_bss»
 9682 |         if (wdev->current_bss) {
      |                 ^~
cc1: некоторые предупреждения считаются ошибками
make[2]: *** [scripts/Makefile.build:250: /home/vitja/work/rtl8821CU/os_dep/linux/ioctl_cfg80211.o] Ошибка 1
make[2]: Цель «/home/vitja/work/rtl8821CU/» не была пересобрана из-за ошибок.
make[1]: *** [Makefile:2012: /home/vitja/work/rtl8821CU] Error 2
make[1]: Target 'modules' not remade because of errors.
make[1]: Leaving directory '/usr/src/linux-6.1.19-gentoo-dist'
make: *** [Makefile:2217: modules] Error 2
make: Target 'module' not remade because of errors.
```

system32uwu, 2023-05-15:
This repo is working fine for me: https://github.com/morrownr/8821cu-20210916, it's up to date

magarto, 2023-07-22:
With this repo 5 GHz is not working for me

> This repo is working fine for me: https://github.com/morrownr/8821cu-20210916, it's up to date

colin-i, 2024-09-26:
I can confirm that morrownr/8821cu-20210916 repo is working fine for a 5 GHz wi-fi usb on a 6.6 kernel.