brektrou/rtl8821CU #220: Error when attempting update from Debian 11 to 12

rwb196884, 2023-06-25:
```
root@calleva:~# apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up linux-image-6.1.0-9-amd64 (6.1.27-1) ...
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-9-amd64.
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KVER=6.1.0-9-amd64.................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-9-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-9-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.1.0-9-amd64 (--configure):
 installed linux-image-6.1.0-9-amd64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-image-amd64:
 linux-image-amd64 depends on linux-image-6.1.0-9-amd64 (= 6.1.27-1); however:
  Package linux-image-6.1.0-9-amd64 is not configured yet.

dpkg: error processing package linux-image-amd64 (--configure):
 dependency problems - leaving unconfigured
Setting up linux-headers-6.1.0-9-amd64 (6.1.27-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-9-amd64.
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KVER=6.1.0-9-amd64.................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-9-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-9-amd64 failed!
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/header_postinst.d at /var/lib/dpkg/info/linux-headers-6.1.0-9-amd64.postinst line 11.
dpkg: error processing package linux-headers-6.1.0-9-amd64 (--configure):
 installed linux-headers-6.1.0-9-amd64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-headers-amd64:
 linux-headers-amd64 depends on linux-headers-6.1.0-9-amd64 (= 6.1.27-1); however:
  Package linux-headers-6.1.0-9-amd64 is not configured yet.

dpkg: error processing package linux-headers-amd64 (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 linux-image-6.1.0-9-amd64
 linux-image-amd64
 linux-headers-6.1.0-9-amd64
 linux-headers-amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
r
```
and
```
root@calleva:~# cat  /var/lib/dkms/rtl8821CU/5.4.1/build/make.log
DKMS make.log for rtl8821CU-5.4.1 for kernel 6.1.0-9-amd64 (x86_64)
Sun 25 Jun 11:40:55 BST 2023
[... 54 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2494:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2494 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:925:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  925 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2618:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2618 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-9-common/scripts/Makefile.build:255: /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-9-common/Makefile:2037: /var/lib/dkms/rtl8821CU/5.4.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-9-amd64'
make: *** [Makefile:2217: modules] Error 2
root@calleva:~#
```

kelebek333, 2023-06-26:
You can try to use rtw88 instead of rtl8821cu. https://github.com/lwfinger/rtw88

rwb196884, 2023-06-27:
So: is there a way to remove it so that I can complete the Debian update?
At present this is causing
```
# apt --fix-broken install
```
to fail

kelebek333, 2023-06-27:
Run following commands.

```
sudo dkms remove rtl8821CU/5.4.1 --all
sudo apt install -f
```

Then You can install rtw88.

rwb196884, 2023-06-27:
Thank you <3 that's got debian updated and restarted.

Strangely, the wireless interface in question has come back up.

closed 2023-07-02 by rwb196884