brektrou/rtl8821CU #229: osdep_service.c:2618:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’?

OlegVladimirov, 2024-05-11:
Hi!
While I run install using dkms there was an error:
```
# ./dkms-install.sh 
About to run dkms install steps...
Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source -> /usr/src/rtl8821CU-5.4.1
Sign command: /lib/modules/6.6.16-current-sunxi/build/scripts/sign-file
Binary /lib/modules/6.6.16-current-sunxi/build/scripts/sign-file not found, modules won't be signed

Building module:
Cleaning build area...
'make' KVER=6.6.16-current-sunxi...................................................................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.16-current-sunxi (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Sign command: /lib/modules/6.6.16-current-sunxi/build/scripts/sign-file
Binary /lib/modules/6.6.16-current-sunxi/build/scripts/sign-file not found, modules won't be signed

Building module:
Cleaning build area...
'make' KVER=6.6.16-current-sunxi.................................................................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.16-current-sunxi (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Finished running dkms install steps.
```

and if using another way of install it's similar error (as in /var/lib/dkms/rtl8821CU/5.4.1/build/make.log):
```
# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.6.16-current-sunxi/build M=/root/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.16-current-sunxi'
  CC [M]  /root/rtl8821CU/core/rtw_cmd.o
  CC [M]  /root/rtl8821CU/core/rtw_security.o
  CC [M]  /root/rtl8821CU/core/rtw_debug.o
  CC [M]  /root/rtl8821CU/core/rtw_io.o
  CC [M]  /root/rtl8821CU/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8821CU/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8821CU/core/rtw_ieee80211.o
  CC [M]  /root/rtl8821CU/core/rtw_mlme.o
  CC [M]  /root/rtl8821CU/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8821CU/core/rtw_mi.o
  CC [M]  /root/rtl8821CU/core/rtw_wlan_util.o
  CC [M]  /root/rtl8821CU/core/rtw_vht.o
  CC [M]  /root/rtl8821CU/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8821CU/core/rtw_rf.o
  CC [M]  /root/rtl8821CU/core/rtw_recv.o
  CC [M]  /root/rtl8821CU/core/rtw_sta_mgt.o
/root/rtl8821CU/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/root/rtl8821CU/core/rtw_sta_mgt.c:379:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  379 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /root/rtl8821CU/include/drv_types.h:107,
                 from /root/rtl8821CU/core/rtw_sta_mgt.c:17:
/root/rtl8821CU/include/sta_info.h:256:17: note: ‘lock’ declared here
  256 |         _lock   lock;
      |                 ^~~~
  CC [M]  /root/rtl8821CU/core/rtw_ap.o
/root/rtl8821CU/core/rtw_ap.c: In function ‘rtw_ap_update_chbw_by_ifbmp’:
/root/rtl8821CU/core/rtw_ap.c:4397:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4397 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
In file included from /root/rtl8821CU/core/rtw_ap.c:17:
/root/rtl8821CU/include/drv_types.h:1020:19: note: ‘padapters’ declared here
 1020 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
/root/rtl8821CU/core/rtw_ap.c:4417:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4417 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
/root/rtl8821CU/include/drv_types.h:1020:19: note: ‘padapters’ declared here
 1020 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
  CC [M]  /root/rtl8821CU/core/mesh/rtw_mesh.o
  CC [M]  /root/rtl8821CU/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/rtl8821CU/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/rtl8821CU/core/rtw_xmit.o
  CC [M]  /root/rtl8821CU/core/rtw_p2p.o
  CC [M]  /root/rtl8821CU/core/rtw_rson.o
  CC [M]  /root/rtl8821CU/core/rtw_tdls.o
  CC [M]  /root/rtl8821CU/core/rtw_br_ext.o
  CC [M]  /root/rtl8821CU/core/rtw_iol.o
  CC [M]  /root/rtl8821CU/core/rtw_sreset.o
  CC [M]  /root/rtl8821CU/core/rtw_btcoex_wifionly.o
  CC [M]  /root/rtl8821CU/core/rtw_btcoex.o
  CC [M]  /root/rtl8821CU/core/rtw_beamforming.o
  CC [M]  /root/rtl8821CU/core/rtw_odm.o
  CC [M]  /root/rtl8821CU/core/rtw_rm.o
  CC [M]  /root/rtl8821CU/core/rtw_rm_fsm.o
  CC [M]  /root/rtl8821CU/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8821CU/os_dep/osdep_service.o
/root/rtl8821CU/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/root/rtl8821CU/os_dep/osdep_service.c:2494:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2494 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/root/rtl8821CU/os_dep/osdep_service.c:925:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  925 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/root/rtl8821CU/os_dep/osdep_service.c: In function ‘rtw_random32’:
/root/rtl8821CU/os_dep/osdep_service.c:2618:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2618 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /root/rtl8821CU/os_dep/osdep_service.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.16-current-sunxi/Makefile:1915: /root/rtl8821CU] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.16-current-sunxi'
make: *** [Makefile:2217: modules] Error 2

```

Jan64X, 2024-05-11:
same error here

Xenon03, 2024-06-04:
I'm getting the same :(

skibon02, 2024-07-28:
same error, trying to upgrade to mint 22 from 21.3

jaum20, 2024-07-28:
same error on Debian bullseye

urpwn3d, 2024-08-10:
Same error on Linux Mint 22 (Kernel 6.8.0-40-generic).

thematrix999, 2024-10-17:
Same error.
Debian 12
kernel 6.1.0-26-amd64

hymie0, 2024-11-04:
> I'm getting the same :( also, by changing prandom_u32 to get_random_u32 another error appears: ![Screenshot from 2024-06-04 19-52-07](https://private-user-images.githubusercontent.com/46632334/336556745-48f13e39-a124-40d3-bf66-765269c823c1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3MzI0MTAsIm5iZiI6MTczMDczMjExMCwicGF0aCI6Ii80NjYzMjMzNC8zMzY1NTY3NDUtNDhmMTNlMzktYTEyNC00MGQzLWJmNjYtNzY1MjY5YzgyM2MxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA0VDE0NTUxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiNDgzNWM5YWY1MzFhOWFhZDAzOTdlNjA1NjdiZGY0N2M4Y2MyZDI1YWQ3MTY2MGM2ZWQwNzRhNjhkNGI2NDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MLpsW3fecsfvzJPf2_imE1WI52LxkVPe_5LdbpHw14w)

I'm pretty sure that just means "You fixed that error and went on to the next error.

I'm pretty sure that replacing prandom_u32 with get_random_u32 is the correct answer, see also
https://patchew.org/linux/20221010230613.1076905-1-Jason@zx2c4.com/20221010230613.1076905-6-Jason@zx2c4.com/

hymie0, 2024-11-04:
I hate to plug a different repo, but https://github.com/morrownr/8821cu-20210916 is working for me.